bakerstu/openmrn #18: better Makefile support for generated cdi.xml

balazsracz, 2015-07-05:
In the io_board Makefiles we use a generated cdi file. It's called cdi.nxml and there are custom commands to link it into the binary.

Two TODO items here:
- the name should change: cdi.nxml is the wrong name, it should be cdi.xmlout
- we should probably have first-class support in prog.mk for generating the CDI so that the application target directory does not need a customized Makefile.

balazsracz, 2015-07-05:
fyi I tried to make the name be "cdi.xml" but failed due to a strange Makefile dependency issue.

- if there is no cdi.xml before the build, then we won't figure out that a cdi.xml needs to be built, compiled into an .o file and linked into the binary. The effect is that you need to do make twice before the cdi is build and linked into the binary (ugly). If I fix that however, then:
- if there is a cdi.xml before the build then we'll add cdi.o twice to the link step and cause a duplicate symbol linker error.

nambabwe, 2015-07-06:
We also need to update the "make clean" to "rm

nambabwe, 2015-07-06:
We also need to update the "make clean" to "rm -f cdi.nxml ..." instead of "rm -f cdi.xml ..."

balazsracz, 2016-08-02:
This is completed; the standard prog.mk now has provisions to detect whether a config XML needs to be generated (by the presence of a "config.hxx" with some magic words), generates, compiles, and links the config XML to the binary.

Closing issue.

closed 2016-08-02 by balazsracz