bakerstu/openmrn #20: set search order consistently in etc/path.mk

kphannan, 2015-07-06:
The search order specified for directories in ```etc/path.mk``` is inconsistent.  Some searches have **$(HOME).....** as a first entry, while others have it last.  When used **$(HOME)** should have the highest precedence which would allow the user to provide a local override for any default system directory.

for example:
```
################ tivaware ##################
ifndef TIVAWAREPATH
SEARCHPATH := \
  /opt/ti/TivaWare/default \
  /opt/TivaWare/default \
  /opt/TivaWare \
  $(HOME)/TivaWare
```

should be
```
################ tivaware ##################
ifndef TIVAWAREPATH
SEARCHPATH := \
  $(HOME)/TivaWare \
  /opt/ti/TivaWare/default \
  /opt/TivaWare/default \
  /opt/TivaWare
```

kphannan, 2015-07-06:
I'm looking at these as I build the VM setup with Packer and SaltStack.  I can make the changes as I find them.

bakerstu, 2015-07-06:
Kevin,
I agree, having better consistency here is a good idea.  Feel free to make changes as you see fit.