bakerstu/openmrn #25: Verify Node ID Global with Node ID Payload

bakerstu, 2016-10-01:
The Verify Node ID Global message (as described in the Message Network S & TN) is allowed to contain a Node ID in the payload of the message.  If the Node ID is present, only a node with a matching Node ID is supposed to respond.  It looks as though in our case, all of our nodes respond regardless as to whether the Node ID is present.  The Node ID payload does not appear to be validated in OpenMRN.

I've verified that the RailStars and RR-CirKits nodes do not have this issue.

Here is the Python test trace:
```
   connect to  localhost : 12021
   send    :X19490AAAN020199FF001E;
   receive :X191705F9N020199FF001E;
   send    :X19490AAAN;
   receive :X19170149N02015700003E;
   receive :X19170060N050201020211;
   receive :X19170065N050201020214;
   receive :X191705F9N020199FF001E;
   send    :X19490AAAN000000000001;
   receive :X191702A8N060100000002;
Global verify with wrong node ID should not receive reply but did:  :X19170149N02015700003E;
```

bakerstu, 2016-10-01:
Ignore the message "Global verify with wrong node ID should not receive reply but did:  :X19170149N02015700003E;".  This is a bug in the print statement of the Python test.  The issue still exists.

balazsracz, 2016-10-03:
I think there might be a race condition in the test that produced the above packet dump.

Depending on what the exact timing of the packets is the above capture is standards compliant:
receive :X191702A8N060100000002;  (last line) is a valid response to the request
send    :X19490AAAN;  (third line).

If the test script does not put in an appropriately long wait for the bus to quiesce between the request on line 3 and the request on line 8, then we have a race condition that could trigger a rearrangement of the packets on the bus and a misattribution of a response triggering a false warning.

We do have a unit test the verifies that nodes do not respond to global verify node ID requests with mismatching node IDs:
https://github.com/bakerstu/openmrn/blob/16132b1cd7b701e5294b19b02c5c5772aef9664f/src/nmranet/IfImpl.cxxtest#L48

I recommend to proceed as follows:
- add a longer wait in the test script between the global and the addressed requests
- if this makes the warning go away, we can close this bug.
- else print the timestamps of the packets in the testing framework and let me know how to reproduce the test setup.

Note: the request with the address should have a valid node ID as address; any node ID starting with 00 is invalid as per the standard, so I recommend updating the test slightly:
:X19490AAAN010000000001;

bakerstu, 2016-10-03:
Balazs,

This is indeed the issue.  I updated the test to pause for 3 seconds and then flush the receive message buffer before proceeding.  This has resolved the issue, no fix required for OpenMRN.

closed 2016-10-03 by bakerstu