bakerstu/openmrn #26: Event Identify Range not produced on Identify Events Global

bakerstu, 2016-10-02:
Global:

```
   connect to  localhost : 12021
   send    :X19970AAAN;
   receive :X19547065N0502010202140010;
   receive :X19547060N0502010202000000;
   receive :X19547060N0502010202000001;
   receive :X19547060N0502010202000002;
   receive :X19547060N0502010202000003;
   receive :X19547060N0502010202000004;
   receive :X195472A8N0101000000000303;
   receive :X19547060N0502010202000005;
   receive :X19547060N0502010202000006;
   receive :X19547060N0502010202000007;
   receive :X19547060N0502010202000008;
   receive :X19547060N0502010202000009;
   receive :X19547060N050201020200000A;
   receive :X19547060N050201020200000B;
   receive :X19547F57N0101000000000303;
   receive :X19547060N050201020200000C;
   receive :X19547060N050201020200000D;
   receive :X19547060N050201020200000E;
   receive :X19547060N050201020200000F;
   receive :X194C7060N0502010202000000;
   receive :X194C7060N0502010202000001;
   receive :X194C7060N0502010202000002;
   receive :X19547C06N0101000000000303;
   receive :X194C7060N0502010202000003;
   receive :X194C7060N0502010202000004;
   receive :X194C7060N0502010202000005;
   receive :X194C7060N0502010202000006;
   receive :X194C7060N0502010202000007;
   receive :X194C7060N0502010202000008;
   receive :X194C7060N0502010202000009;
   receive :X194C55F9N010000000000FFFE;
   receive :X194C45F9N010000000000FFFF;
   receive :X194C7060N050201020200000A;
   receive :X194C7060N050201020200000B;
   receive :X194C7060N050201020200000C;
   receive :X195B45F9N0502010202000000;
   receive :X194C7060N050201020200000D;
   receive :X194C7060N0201570000400006;
   receive :X194C7060N0201570000400007;
   receive :X19547065N0502010202140011;
   receive :X195445F9N010000000000FFFF;
   receive :X195455F9N010000000000FFFE;
   receive :X194C45F9N010000000000FFFF;
   receive :X194C55F9N010000000000FFFE;
   receive :X19547065N0502010202140012;
   receive :X19547065N0502010202140013;
   receive :X19547065N0502010202140014;
   receive :X19547065N0502010202140015;
   receive :X19547065N0502010202140016;
   receive :X19547065N0502010202140017;
   receive :X19547065N0502010202140018;
   receive :X195445F9N0502010202000000;
   receive :X195455F9N0502010202000001;
   receive :X19547065N0502010202140019;
   receive :X19547065N050201020214001A;
   receive :X19547065N050201020214001B;
   receive :X19547065N050201020214001C;
   receive :X19547065N050201020214001D;
   receive :X19547065N050201020214001E;
   receive :X19547065N050201020214001F;
   receive :X194C7065N0201570000400006;
   receive :X194C7065N0201570000400012;
   receive :X195242A8N090099FF00000000;
   receive :X194C7065N0502010202140022;
   receive :X194C7065N0502010202140023;
   receive :X194C7065N0502010202140024;
   receive :X19524F57N090099FF00000000;
   receive :X194C7065N0502010202140025;
   receive :X194C7065N0502010202140026;
   receive :X194C7065N0502010202140027;
   receive :X19524C06N090099FF00000000;
   receive :X194C7065N0502010202140028;
   receive :X194C7065N0502010202140029;
   receive :X194C7065N050201020214002A;
   receive :X194C7065N050201020214002B;
   receive :X194C7065N050201020214002C;
   receive :X194C7065N050201020214002D;
   receive :X194C7065N050201020214002E;
   receive :X194C7065N050201020214002F;
   receive :X19547149N02015700003E0006;
   receive :X19547149N02015700003E0007;
   receive :X19547149N02015700003E0012;
   receive :X19547149N02015700003E0013;
   receive :X19547149N02015700003E001E;
   receive :X19547149N02015700003E001F;
   receive :X19547149N02015700003E002A;
   receive :X19547149N02015700003E002B;
   receive :X19547149N02015700003E0036;
   receive :X19547149N02015700003E0037;
   receive :X19547149N02015700003E0042;
   receive :X19547149N02015700003E0043;
   receive :X19547149N02015700003E004E;
   receive :X19547149N02015700003E004F;
   receive :X19547149N02015700003E005A;
   receive :X19547149N02015700003E005B;
   receive :X19547149N02015700003E0066;
   receive :X19547149N02015700003E0067;
   receive :X19547149N02015700003E0072;
   receive :X19547149N02015700003E0073;
   receive :X19547149N02015700003E007E;
   receive :X19547149N02015700003E007F;
   receive :X19547149N02015700003E008A;
   receive :X19547149N02015700003E008B;
   receive :X19547149N02015700003E0096;
   receive :X19547149N02015700003E0097;
   receive :X19547149N02015700003E00A2;
   receive :X19547149N02015700003E00A3;
   receive :X19547149N02015700003E00AE;
   receive :X19547149N02015700003E00AF;
   receive :X19547149N02015700003E00BA;
   receive :X19547149N02015700003E00BB;
   receive :X195B45F9N0502010202000001;
<none>
Timeout
  Found 4 consumer events
  Found 4 producer events
```

Addressed:

```
   connect to  localhost : 12021
   send    :X19968AAAN05F9;
   receive :X194C55F9N010000000000FFFE;
   receive :X194C45F9N010000000000FFFF;
   receive :X195445F9N010000000000FFFF;
   receive :X195455F9N010000000000FFFE;
   receive :X194C45F9N010000000000FFFF;
   receive :X194C55F9N010000000000FFFE;
   receive :X195445F9N0502010202000000;
   receive :X195455F9N0502010202000001;
   receive :X194A45F9N090099FF00000000;
   receive :X195B45F9N0502010202000001;
<none>
Timeout
  Found 4 consumer events
  Found 4 producer events
  Found Consumer Range 90099ff00000000 - 90099ffffffffff
```

balazsracz, 2016-10-02:
Could you post reproduction information?
Specifically: what node (where is the code) failed to output the consumer
range identified message. This responsibility is in the event handler, so
I'd need to know which event handler to check.

--
Sent from my mobile device

bakerstu, 2016-10-03:
Yes, I'll collect this info and get back to you.