bakerstu/openmrn #29: Should we rename nmranet in the codebase?

balazsracz, 2016-12-11:
The name 'nmranet' was sunset two years ago by the NMRA.

We have extensive occurrences in our codebase -- a directory 'nmranet', a namespace 'nmranet' and the core buffer type called NMRANetMessage.

Should we rename these? If yes, rename to what? Reasonable options are:
- LCC
- OpenLCB

This is probably a massive undertaking that has to span multiple repositories in one go.

[please add your comments to this issue thread]

bakerstu, 2016-12-11:
I agree that we should make this change.  I would suggest that we should use OpenLCB with the "OpenLCB" case sensitivity for object and file naming and "openlcb" for the the namespace.  I prefer OpenLCB over lcc for two reasons:

1. OpenLCB covers the super-set of features beyond the LCC standards
2. OpenLCB is the same number of characters as NMRAnet, making the existing spacing the same as the new spacing for formatting purposes.

This is a huge change.  Timing is critical.

balazsracz, 2016-12-11:
I'm fine with the OpenLCB prefix and object names. None of the filenames contain NMRAnet today (seems that was a large scale change that we already completed a long while ago), so I guess that will be the same with OpenLCB too.

Can we have the directory name with lowercase though, just like the namespace?
`src/openlcb/*`
and
`namespace openlcb {...}`

There is one major source of NMRAnet naming in the codebase (just grepped it):
NMRAnetMessage is a class which we use a lot of places, most typically in the form of 
`Buffer<NMRAnetMessage>.` Should this be
- `Buffer<OpenLCBMessage>` / `Buffer<openlcb::OpenLCBMessage>` or 
- `Buffer<Message>` / `Buffer<openlcb::Message>` ?
when called in or out of the openlcb namespace, respectively.

I don't quite like either of them; `openlcb::OpenLCBMessage` seems pretty long and redundant (even though it's the sane # of characters as `nmranet::NMRAnetMessage`) but `Message` is a way too generic name for a class even within the `openlcb` namespace.

bakerstu, 2016-12-11:
I generally prefer all lowercase directory names (sometimes with separating underscores), and this is the way it is today with nmranet, so no problem there.

I do not like Buffer<OpenLCBMessage> if it is already in namespace openlcb, it is redundant.  You are correct though that message is quite generic.  This is openlcb general message format, so perhaps something like "GenMessage" or "GeneralMessage" would be appropriate.

dpharris, 2016-12-11:
> I do not like Buffer if it is already in namespace openlcb, it is

I defer to your choices, but agree changing from nmranet is overdue.
David

balazsracz, 2016-12-16:
This is now done.

closed 2016-12-16 by balazsracz