bakerstu/openmrn #48: Incorrect NMRAnetNode.hxx, NMRAnetMessageID.hxx, NMRAnetIf.hxx in Stream.[hc]xx

evili, 2017-01-26:
Hi,

There are still three includes that reference obsoleted files. In openlcb/Stream.cxx openlcb/Stream.hxx you can still find the following:

`
$ grep -n  NMRAnet src/openlcb/* | grep include 

src/openlcb/Stream.cxx:35:#include "openlcb/NMRAnetNode.hxx"

src/openlcb/Stream.cxx:36:#include "openlcb/NMRAnetMessageID.hxx"

src/openlcb/Stream.hxx:37:#include "openlcb/NMRAnetIf.hxx"

`
These files are missing from the repo since 2014 at commits: 

[316e43f08](https://github.com/bakerstu/openmrn/commit/316e43f08542aaddfcc5db634c736c6b8ee6544c)

balazsracz, 2017-01-26:
Yeah, Stream.hxx and Stream.cxx is not functional at the moment -- in fact
they have never been integrated into the stack's asynchronous execution
infrastructure. It probably doesn't reflect the current draft Standard for
streams either.

We should probably dust it off and actually implement streams handling; it
would also significantly reduce the complexity of the BootloaderClient,
which currently does the expected stuff barefoot without help from the
stack.

Do you have a need / use-case of streams?

Balazs

evili, 2017-01-26:
Hi, 

I do not use Streams.  I was just trying to compile with eclipse. 

Thanks, 
  Evili