bakerstu/openmrn #72: remove cJSON and tinyxml2 libraries

balazsracz, 2017-06-22:
these are not used at all.

bakerstu, 2017-06-22:
I'm in favor of getting rid of them (unless JohnSL needs C JSON).  Then again, including these as part of the OpenMRN package goes against the grain of providing dependent packages separately so that we are not redistributing someone else copyrighted material.

balazsracz, 2017-06-22:
@JohnSL do you need that specific JSON library? Or would you prefer to pick your own, or upgrade and move to the same distribution policy as all other libraries (i.e. external in /opt).

JohnSL, 2017-06-24:
I agree with @bakerstu that they shouldn't be in OpenMRN. I will be using JSON for my own application, and can include the library there.