bakerstu/openmrn #244: Esp32: node reboot from jmri doesn't reboot the nodr

atanisoft, 2019-03-03:
When in the CDI pane for an esp32 node clicking more then reboot does nothing.

Most likely this just needs a platform specific hook to be put in place but I'm unsure where exactly to put it.

balazsracz, 2019-03-04:
This is done by an extern "C" function called `reboot()`
https://github.com/bakerstu/openmrn/blob/master/boards/ti-bracz-acc3/HwInit.cxx#L111

balazsracz, 2019-03-04:
If there is a standard arduino call to reboot the MCU, then we could put that function into OpenMRNLite.cpp.

atanisoft, 2019-03-04:
I'm not aware of a generic reboot function for all Arduino platforms, on the ESP32 it can be done through a couple approaches: ESP.restart() or esp_restart() (IDF). I'm leaning towards the IDF function since that is what the ESP.restart() calls and we would avoid an include of Arduino.h.

balazsracz, 2019-03-04:
sounds like it needs to be conditionally compiled for the esp32 then.
I don't see a problem in including Arduino.h into OpenMRNLite.cpp. That does not poison anybody's namespace.

atanisoft, 2019-03-04:
will namespace play any factor for the C function?

balazsracz, 2019-03-04:
instead of a namespace you have to wrap it in `extern "C" {` like here:
https://github.com/bakerstu/openmrn/blob/master/boards/ti-bracz-acc3/HwInit.cxx#L89

atanisoft, 2019-03-04:
Spotted the need for the extern "C" block and tested within the openmrn_arduino namespace and it works as expected now. See PR #247.

commit f23b9d23 (2019-03-04): Fix #244 (reboot node on ESP32)

Adding definition for reboot specific to the ESP32 so that when the node
requires a reboot (config update) or is requested to reboot via JMRI (or
other means) it will now be performed as expected.
commit 477ddfb6 (2019-03-16): Merge branch 'master' of github.com:bakerstu/openmrn into bracz-js-client-expand

* 'master' of github.com:bakerstu/openmrn: (30 commits)
  Added 50ms delay after export to give the kernel time to change PWM directory ownership. (#250)
  Simple, portable LinuxPWM using sysfs API (#232)
  Separates arduino README.md into three files: - README.md for users - CONTRIBUTING.md for developers - RELEASE.md for maintainers
  Fix #244 (reboot node on ESP32)
  Include errno.h since it fails to compile on Arduino without this.
  remove reference to include as it is not necessary now.
  Updates for Arduin [... cut]
closed 2019-03-04 by balazsracz