bakerstu/openmrn #398: Config file gets opened twice in SimpleStack

balazsracz, 2020-07-09:
In SimpleStack, the configUpdateFlow_ opens the config file and holds on to an fd to it. This fd is also accessible by the application writer, as it is passed in as arguments to the ConfigUpdateListeners and also accessible via direct API in appl_main.

Then there is a FileMemorySpace which receives writes from the LCC bus. This object opens the file a second time, and has a different fd.
There is no correctness problem AFAIK, because the filesystem is supposed to make data written through one fd available to read through another fd immediately. Nevertheless, we should merge these two and have only one fd that points to the config file.
The prime use-case is that upon UPDATE_COMPLETE or REBOOT or FREEZE we should fdflush() the fd on which we were writing.

atanisoft, 2020-07-09:
it looks like potentially three FDs are allocated: SPACE_ACDI_USR, SPACE_CONFIG, configUpdateFlow_.

atanisoft, 2020-07-09:
possibly switch from FileMemorySpace(char *, address_t) to FileMemorySpace(int, address_t)?

atanisoft, 2020-07-09:
it would potentially also raise the concern of concurrent reads from a single fd (due to seek, etc)

atanisoft, 2020-11-08:
@balazsracz found one case where this behavior is breaking. LittleFS appears to be caching file data on a per-handle basis and does not keep the caches in sync. LittleFS will be an option in arduino-esp32 v2.0.0. I'm investigating if we can disable or at least tune the caching so this doesn't cause as many problems.

Is there any quick fix that we can apply to OpenMRN to remove the multiple open files so that there is only one handle to work with between all flows?

balazsracz, 2020-11-08:
This is not really a quick fix, but here is what we should do:
- First we need to do some background work in reviewing the usage of the different fd variables. Specifically we can only use the same fd in different places if we never rely on the file offset remaining the same between consecutive operations. In other words everywhere we do a read or write we have to do an lseek first. Asynchronous operations are not ok (O_NONBLOCK).
- in SimpleStack add a variable that holds the config file fd, initialized to -1.
- in simplestack, where we create or open the config file (and possibly do a factory reset), hold on to the fd.
- then where we instantiate the filememoryspace objects, instead of checking if CONFIG_FILE is non null, we should check if we have an fd for the config file. Then use the fd to create the object instead of the path.

For memory spaces this is probably doable. A hard one is the SNIP handler but maybe we can leave that as is.

atanisoft, 2020-11-08:
Since SNIP doesn't typically change I think it may be ok to leave it.

Another thing to consider is in the MemoryConfig flow is when we receive "update complete" is we call fsync on the FD before invoking apply_configuration on listeners. This could be an optional callback since it not needed on all FS types, something similar to the activity callback on If.

balazsracz, 2020-11-08:
does fsync break for you?
If it doesn't break I see no reason to add the complexity to make it
optional.

atanisoft, 2020-11-08:
I haven't seen any issues with calling it yet and in fact have needed to add `AutoSyncFileFlow` when using LittleFS and SD as storage but not when using SPIFFS.

atanisoft, 2020-11-09:
I opened https://github.com/joltwallet/esp_littlefs/issues/18 to track the odd behavior and provided a very basic test case which reproduces the behavior seen in https://github.com/atanisoft/esp32olcbhub/issues/7. https://github.com/littlefs-project/littlefs/issues/483 has now been filed upstream to track what appears to be a cache invalidation bug occurring after `fsync` is called.

I'm thinking we may not need another variable for holding the fd since we have `configUpdateFlow_` which tracks the handle passed to various consumers via apply_configuration. I think we can call `configUpdateFlow_.open_file()` from `SimpleStackBase::default_start_node()` but it will need to be verified for async (O_NONBLOCK) which is used by `FileMemorySpace` when `__FreeRTOS__` is defined.

commit 488e6f1b (2021-07-25): Reduce the number of times the config file is opened in SimpleStack (#555)

Fixes  #398.

* Do not permit repeated calls to ConfigUpdateFlow::open_file() as this was ignoring the path argument.

* Add ConfigUpdateFlow::get_fd() for callers which need access to the shared config file descriptor.

* Reuse the file descriptor for the config file that was opened by the ConfigUpdateFlow when creating FileMemorySpace instances in SimpleStackBase.

* Add missing #ifdef GTEST and fix #endif comments.
closed 2021-07-25 by balazsracz