bakerstu/openmrn #399: Errors compiling applications/hub on OSX

dpharris, 2020-07-10:
I have been having problems compiling **applications/hub** on **OSX**.  

### 1.  src/utils/FileUtils.cxx
```
/Users/dpharris/openmrn/src/utils/FileUtils.cxx:111:12: error: comparison of unsigned expression < 0 is always false
      [-Werror,-Wtautological-compare]
    if (nr < 0) {
        ~~ ^ ~
1 error generated.
make[4]: *** [FileUtils.o] Error 1
make[3]: *** [build-utils] Error 2
make[2]: *** [/Users/dpharris/openmrn/targets/mach.x86/lib/timestamp] Error 2
make[1]: *** [build-mach.x86] Error 2
make: *** [build-targets] Error 2
```

- nr is defined as "size_t nr".  I made a local fix by commenting out the offending if-block.  

### 2. src/utils/SocketClient.hxx and .cxx ....
```
/Users/dpharris/openmrn/src/utils/SocketClient.hxx:630:28: error: implicit instantiation of undefined template 'std::__1::array<SocketClient::Attempt, 5>'
    std::array<Attempt, 5> strategyConfig_{{
                           ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/__tuple:95:65: note: template is declared here
template <class _Tp, size_t _Size> struct _LIBCPP_TYPE_VIS_ONLY array;
                                                                ^
In file included from /Users/dpharris/openmrn/src/utils/SocketClient.cxx:41:
/Users/dpharris/openmrn/src/utils/SocketClient.hxx:615:26: error: no matching constructor for initialization of 'StateFlowBase::StateFlowTimer'
    StateFlowTimer timer_{this};
                         ^~~~~~
/Users/dpharris/openmrn/src/executor/StateFlow.hxx:461:11: note: candidate constructor (the implicit copy constructor) not viable: no known conversion from 'SocketClient *' to 'const StateFlowBase::StateFlowTimer' for 1st argument
    class StateFlowTimer : public ::Timer
          ^
/Users/dpharris/openmrn/src/executor/StateFlow.hxx:465:9: note: candidate constructor not viable: no known conversion from 'SocketClient *' to 'StateFlowBase *' for 1st argument
        StateFlowTimer(StateFlowBase *parent)
        ^
In file included from /Users/dpharris/openmrn/src/utils/SocketClient.cxx:41:
/Users/dpharris/openmrn/src/utils/SocketClient.hxx:98:9: error: cannot initialize object parameter of type 'StateFlowBase' with an expression of type 'SocketClient'
        start_flow(STATE(start_connection));
        ^~~~~~~~~~
/Users/dpharris/openmrn/src/utils/SocketClient.hxx:164:26: error: no matching constructor for initialization of 'AtomicHolder'
            AtomicHolder h(this);
                         ^ ~~~~
/Users/dpharris/openmrn/src/utils/Atomic.hxx:163:7: note: candidate constructor (the implicit copy constructor) not viable: no known conversion from 'SocketClient *' to 'const AtomicHolder' for 1st argument
class AtomicHolder
      ^
/Users/dpharris/openmrn/src/utils/Atomic.hxx:170:5: note: candidate constructor not viable: no known conversion from 'SocketClient *' to 'Atomic *' for 1st argument
    AtomicHolder(Atomic *parent)
    ^
In file included from /Users/dpharris/openmrn/src/utils/SocketClient.cxx:41:
/Users/dpharris/openmrn/src/utils/SocketClient.hxx:170:17: error: cannot initialize object parameter of type 'StateFlowBase' with an expression of type 'SocketClient'
        while (!is_terminated())
                ^~~~~~~~~~~~~
/Users/dpharris/openmrn/src/utils/SocketClient.hxx:179:22: error: no matching constructor for initialization of 'AtomicHolder'
        AtomicHolder h(this);
                     ^ ~~~~
/Users/dpharris/openmrn/src/utils/Atomic.hxx:163:7: note: candidate constructor (the implicit copy constructor) not viable: no known conversion from 'SocketClient *' to 'const AtomicHolder' for 1st argument
class AtomicHolder
      ^
/Users/dpharris/openmrn/src/utils/Atomic.hxx:170:5: note: candidate constructor not viable: no known conversion from 'SocketClient *' to 'Atomic *' for 1st argument
    AtomicHolder(Atomic *parent)
    ^
In file included from /Users/dpharris/openmrn/src/utils/SocketClient.cxx:41:
/Users/dpharris/openmrn/src/utils/SocketClient.hxx:186:16: error: cannot initialize object parameter of type 'StateFlowBase' with an expression of type 'SocketClient'
        return is_terminated();
               ^~~~~~~~~~~~~
/Users/dpharris/openmrn/src/utils/SocketClient.hxx:199:26: error: no matching constructor for initialization of 'AtomicHolder'
            AtomicHolder h(this);
                         ^ ~~~~
/Users/dpharris/openmrn/src/utils/Atomic.hxx:163:7: note: candidate constructor (the implicit copy constructor) not viable: no known conversion from 'SocketClient *' to 'const AtomicHolder' for 1st argument
[... 52 lines cut ...]
class AtomicHolder
      ^
/Users/dpharris/openmrn/src/utils/Atomic.hxx:170:5: note: candidate constructor not viable: no known conversion from 'SocketClient *' to 'Atomic *' for 1st argument
    AtomicHolder(Atomic *parent)
    ^
In file included from /Users/dpharris/openmrn/src/utils/SocketClient.cxx:41:
/Users/dpharris/openmrn/src/utils/SocketClient.hxx:459:24: error: cannot initialize object parameter of type 'StateFlowBase' with an expression of type 'SocketClient'
                return exit();
                       ^~~~
/Users/dpharris/openmrn/src/utils/SocketClient.hxx:463:9: error: no matching function for call to object of type 'std::function<void (int, Notifiable *)>'
        callback_(fd_, this);
        ^~~~~~~~~
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/functional:1541:9: note: candidate function not viable: no known conversion from 'SocketClient *' to 'Notifiable *' for 2nd
      argument
    _Rp operator()(_ArgTypes...) const;
        ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[2]: *** [SocketClient.o] Error 1
make[1]: *** [build-utils] Error 2
make: *** [/Users/dpharris/openmrn/targets/mach.x86/lib/timestamp] Error 2
```

Thanks for any help.
David

balazsracz, 2020-07-10:
The first one needs to be fixed by changing it to `ssize_t` from `size_t`.

The second is a stream of errors going back to one root cause, so looking at the errors indidivually makes no sense beyond the first one.
Could you try adding `#include <array>` to the top of SocketClient.hxx?

dpharris, 2020-07-10:
1. Fixed.
2. Improved, but now: 
In file included from /Users/dpharris/openmrn/src/openlcb/TractionCvSpace.cxx:37:
In file included from /Users/dpharris/openmrn/src/dcc/ProgrammingTrackBackend.hxx:41:
/Users/dpharris/openmrn/src/dcc/DccOutput.hxx:189:17: error: no template named 'atomic_uint8_t' in
      namespace 'std'; did you mean 'atomic_init'?
    static std::atomic_uint8_t outputDisableReasons_;
           ~~~~~^~~~~~~~~~~~~~
                atomic_init

dpharris, 2020-07-27:
I think I fixed this by;
replacing `std::atomic_uint8_t`
with: `std::atomic<uint8_t>`
at four (4) places in **src/Dcc/DccOutput.hxx**

I assume that this means that OSX does not have these definitions, or perhaps I am behind as I am on OSX 10.10.5.

balazsracz, 2020-07-27:
I fixed that 11 days ago with
https://github.com/bakerstu/openmrn/commit/771e517887d9c0bd6fed8a49ba42032b554797c6
could you check if master compiles for you now?

dpharris, 2020-07-27:
Hi --

I did a pull before I tried today.  -- BUT I wasn't in master.
Checked out master, and did another pull.

Now error:

g++ -c -g -O0 -Wall -Werror -MD -MP -std=c++0x -m32 -fno-stack-protector
-D_GNU_SOURCE -D__STDC_FORMAT_MACROS -I./ -I/Users/dpharris/openmrn/src/
-I/Users/dpharris/openmrn/include -I/Users/dpharris/openmrn/include/mach
-MD -MF Console.d /Users/dpharris/openmrn/src/console/Console.cxx -o
Console.o

In file included from /Users/dpharris/openmrn/src/console/Console.cxx:38:

In file included from /Users/dpharris/openmrn/src/console/Console.hxx:39:

In file included from /Users/dpharris/openmrn/src/executor/Service.hxx:38:

In file included from /Users/dpharris/openmrn/src/executor/Executor.hxx:51:

*/Users/dpharris/openmrn/src/os/OSSelectWakeup.hxx:213:15: **error: **no
member named 'pselect' in the global namespace*

            ::pselect(nfds, readfds, writefds, exceptfds, &timeout,
&origMask_);

*            ~~^*

1 error generated.

David

balazsracz, 2020-07-27:
hm that's harder. There might be a feature test macro needed for pselect to
be visible.
This depends on the libc that your system ships with so I can't reproduce
myself (whereas a compiler i can install).
You could try
man pselect

or search in /use/include for files that declare pselect(...) and see if
there are some #ifdef's around them.

dpharris, 2020-07-27:
Stuart --
I guess you saw the code:

#elif OPENMRN_HAVE_PSELECT
        struct timespec timeout;
        timeout.tv_sec = deadline_nsec / 1000000000;
        timeout.tv_nsec = deadline_nsec % 1000000000;
        int ret =
            ::pselect(nfds, readfds, writefds, exceptfds, &timeout,
&origMask_);
#elif OPENMRN_HAVE_SELECT

OSX appears to have pselect, from man:

*   NAME*

        *pselect* -- synchronous I/O multiplexing a la POSIX.1g

*   LIBRARY*

        Standard C Library (libc, -lc)

*   SYNOPSIS*

        *#include* *<unistd.h>*

Where is  OPENMRN_FEATURE_DEVICE_SELECT defined?

David

atanisoft, 2020-07-27:
@dpharris include/openmrn_features.h

dpharris, 2020-07-28:
It has that.

#include <unistd.h>

#include "openmrn_features.h"

#include "utils/Atomic.hxx"

#include "os/os.h"

??? Why did it compile for me before?

I am stuck.

David

bakerstu, 2020-07-28:
David,

I’m looking into it. For the record, I did not want OpenMRN to use pselect(). I wanted to use standard select() with a pipe for asynchronous wakeup. You know, the portable and correct way.

Thanks,
Stuart

dpharris, 2020-07-28:
Thanks guys.

This all came about as I wanted a functional hub, since the one in JMRI
appears to not work well.

Any suggestions on what to use as a hub?  Nucleo, Tiva1294, BBB, Pi, or
stick to my laptop?

Also, it would be nice to get some interest-based routing happening in a
hub, since I am thinking that Wifi connected nodes may become more popular,
and it would be efficient not to send all traffic down their RF links.  Has
anyone thought about  general design and algorithms for that?   Do you
centralize, or let each connection manage itself.  The latter might let
multiprocessor gateways scale better?

Thanks again,
David

Thanks
David

bakerstu, 2020-07-28:
David,

I had to update all my X Code tools and that took a while. Not it is late and I’m going to bed. I’ll keep looking at this later.

Thanks,
Stuart

atanisoft, 2020-07-28:
> I did not want OpenMRN to use pselect().

Out of curiosity, what is the benefit of pselect() over select(), is it just the sigmask? If so, what does overriding the sigmask provide as a benefit?

balazsracz, 2020-07-28:
pselect() is a POSIX standard function and it is portable. MacOS supports
pselect. If we are unable to write code that correctly compiles, that's
entirely our problem.

The benefit of using pselect over regular select is that it can be woken up
if a different thread desires to have the select thread resume its
operation. Making select exit early is important when you need to modify
the fd sets.

bakerstu, 2020-07-28:
So both of these statements are untrue. Yes pselect() is an [optional] POSIX method. I’ve worked with many operating systems over the years that do not support it. Our use of pselect() unnecessarily reduces our portability.

Select can be woken up if a different thread desires it. This is typically done using a pipe.

Thanks,
Stuart

bakerstu, 2020-07-28:
David,

I have submitted a pull request to cleanup the Mac OS X build. It removes 32-bit support and doubles down on 64-bit support.
https://github.com/bakerstu/openmrn/pull/409 <https://github.com/bakerstu/openmrn/pull/409>

Can you pull my branch for the Mac OS X cleanup and see if it works for you? Note that there is not currently mDNS support on Mac OS X, including in the hub. This could be added, it is just time and effort.

It seems very likely that there is a linger race condition with using pselect() on Mac OS X. I have filed an issue about pselect().
https://github.com/bakerstu/openmrn/issues/410 <https://github.com/bakerstu/openmrn/issues/410>

I remember expressing this concern about pselect() at the time the change was made. Here is the commit where we made the change:
https://github.com/bakerstu/openmrn/commit/b09ba703bd80b6ef0e002bec5bbf8e0f3e099fe6 <https://github.com/bakerstu/openmrn/commit/b09ba703bd80b6ef0e002bec5bbf8e0f3e099fe6>

Thanks,
Stuart