bakerstu/openmrn #519: Compiler note in arm Linux -- Debian 10 [Buster]

RobertPHeller, 2021-03-16:
When compiling on a BeagleBone running Debian 10 [Buster], I'm getting these notes:

usr/include/c++/8/bits/stl_tree.h:2146:5: note: parameter passing for argument of type ‘std::_Rb_tree<long long unsigned int, std::pair<const long long unsigned int, openlcb::Node*>, std::_Select1st<std::pair<const long long unsigned int, openlcb::Node*> >, std::less<long long unsigned int>, Allocator<std::pair<const long long unsigned int, openlcb::Node*> > >::const_iterator’ {aka ‘std::_Rb_tree_const_iterator<std::pair<const long long unsigned int, openlcb::Node*> >’} changed in GCC 7.1

Is this something I need to worry about?

balazsracz, 2021-03-16:
no need to worry. This is only a concern if you use more than one compiler version (an old one and a recent one). If you do a make clean, all intermediate artifacts are gone, so everything gets compiled with the current compiler version.

closed 2021-03-16 by balazsracz

RobertPHeller, 2021-03-16:
OK, thanks.