bakerstu/openmrn #711: Possible future compiler problem with MultiConfiguredPC.hxx: std::allocator::construct & std::allocator::destroy going away

RobertPHeller, 2023-04-29:
When compiling with -std=gnu++20 (the default for ESP-IDF-V5.0.1), MultiConfiguredPC.hxx gets (fatal) compiler errors involving the use of std::allocator::construct & std::allocator::destroy.  Not a problem now if -std=gnu++14, but will probably be a problem someday...

bakerstu, 2023-04-29:
As a general note, we don't typically like to use the "gnu" language version options. "gnu" tends to pull in components that may be experimental and/or not portable to other non GCC compilers, which we also support. So we prefer -std=c++14 rather than -std=gnu++14.

atanisoft, 2023-04-30:
@bakerstu Unless things have changed, ESP-IDF depends on a few GNU extensions. ESP-IDF defaults to using whichever of these works first: gnu++20 gnu++2a gnu++17 gnu++14

commit 4bc7ab2f (2023-09-23): Fix MultiConfiguredPC.hxx compilation with C++20 (#737)

Due to the C++20 standard removing the construct and destruct methods from std::allocator, it was necessary to adjust the code to use std::allocator_traits instead which has provided these methods since C++11.

Fixes #711

===

* Fix MultiConfiguredPC.hxx compilation with C++20

C++20 removed the construct / destruct methods from std::allocator with the replacement methods being part of std::allocator_traits intead.

Fixes https://github.com/bakerstu/openmrn/issues/711

* Update MultiConfiguredPC.hxx

Fix missing }

* Update MultiConfigu [... cut]
closed 2023-09-23 by balazsracz