bakerstu/openmrn #823: Feature request - Splitting Railcom into Send/Recv part?

chatelao, 2025-02-07:
I like this implementation of RailCom and would like to reuse parts in a decoder implementation.

For this purpose it would be helpful, if the "send" and "receive" part would be a bit more seperate and easy extractable.

I know, this are big whishes, but would maybe also simplifiy the resuse and share of fixes :-).

Thanks a lot for publishing under this great license
Oli

balazsracz, 2025-02-07:
Currently there is very little implemented in terms of the "send" part for railcom.
They are all static functions in the RailcomDefs class here: https://github.com/bakerstu/openmrn/blob/master/src/dcc/RailCom.hxx#L135

There is an example that I started working on but I am not convinced that I ever finished it to the point that it would be working:
https://github.com/bakerstu/openmrn/blob/master/applications/dcc_decoder/main.cxx
Even if it does work, a lot of the code in here would probably need to be refactored into some shared libraries. (Contributions are welcome!)

The good news is that even if you take the files (railcom.hxx/cxx and railcom.h) as-is, the linker will eliminate all code and constants that you are not actually using. So for example the railcom_decode[] array will not take code space in your decoder.

chatelao, 2025-02-07:
Thanks for your feedback and if I will achieve something working based on your start, I'll come back and drop you a note.