bakerstu/openmrn #840: DCC Receiver timing accuracy

balazsracz, 2025-06-07:
https://github.com/bakerstu/openmrn/blob/120ad4ea4b92cd1a461b82ec369fd510c8b8d798/src/dcc/Receiver.hxx#L68

These should be (90, 10000). 
Possibly all the values for DCC_ONE and DCC_ZERO could fail conformance testing if there is any clock inaccuracy.
todo:  allow for up to 2% tolerance.

bakerstu, 2025-06-07:
Something else to add here. I don't see any glitch filtering (filtering out of small pulses). This inherently requires knowledge of the track pin polarity and target edge type (rising or falling). Perhaps it is there but just not obvious to me. Without something like this, we are likely to fail conformance testing.

balazsracz, 2025-06-08:
This code has not been tested with the DCC mobile decoder conformance test
procedure. I am not convinced that it would pass as is.

Specifically for glitches, the important part is that glitches are not
allowed to come in the middle of a bit. They are restricted to a small time
interval around the DCC level transition. Due to this, even in the presence
of a glitchy signal, the sequence of time intervals between zero crossings
only contains one number that matches a valid DCC bit length. That's what
we are looking for in this code. The part we are likely missing is ignoring
the too-short-to-care time intervals, although we do likely have a certain
level of protection purely due to the interrupt latency.